webspiderteam/Bluetooth-DMM-For-Windows #1: ZOYI ZT-300AB pairing error

840922704, 2022-05-19:
[image]
This model is the same as AN9002... But still failed to connect. Anyone can help?

webspiderteam, 2022-05-19:
Thanks for testing :) if it's same as an9002 u dont need to pair. Directly  click on the logo of device it should connect without pairing. but im not sure about the datas are same with an9002. plese inform me about it. if it needs changes i can do.

840922704, 2022-05-19:
@webspiderteam I tested all functions and they work fine! It seems that maybe ZOYI and ANENG use the same manufactory solution.
BTW, is there a way to export the data to file?

webspiderteam, 2022-05-19:
Posibly same but some of them has some decoding differences with same structure. 

There can be a way but device send data 3 or 4 time in a second and file will be too big i think. but if you relly need ican work on it. Can u tell me what kind of data log do u need?
BTW, are you using source code or release pack?

840922704, 2022-05-19:
Just release pack. Actually, I'm not familiar with UWP development. 🤣
A real time chart and file export function are all to do, I think. It's nice if u can add them. ❤️ Maybe after I find some time to figure out how the UWP app works, I'd like to add some code on it.

webspiderteam, 2022-05-19:
It's ok. I dont have much experience with UWP. I'm visual basic 6 user actually but Bluetooth LE needs uwp or wpf. Also there is no popup windows for chart and I didnt wantto make the display bigger also there is no chart control in UWP. There is a solution for it but its in WPF version an i have to convert all stuff to WPF manually.
It would be hard for me to making this kind of change right now. I'm working on it to change code in WPF  but it takes so much time and I dont know when I can finish. I can LOG datas but there is no player to see them again. Maybe importing data to excel and making it a chart can be a solution.

840922704, 2022-05-19:
So I just spent several hours seeking the simpler way. You know how Microsoft treats the UWP and Silverlight.😑
The result is that maybe python is better. For this project, bleak lib brought the cross-platform ability. Here is my code:
`import sys
import platform
import asyncio
import logging
import time

from bleak import BleakClient

logger = logging.getLogger(__name__)

ADDRESS = (
    "fc:58:fa:53:60:86"
)

async def main(address):
    async with BleakClient(address) as client:
        logger.info(f"Connected: {client.is_connected}")
        while 1: 
            value = bytes(await client.read_gatt_char(8, use_cached=1))
            print(value.hex())
            time.sleep(1/3)

if __name__ == "__main__":
    logging.basicConfig(level=logging.INFO)
    asyncio.run(main(sys.argv[1] if len(sys.argv) == 2 else ADDRESS))`
At least what I what is just the data🤣. Further, using HTML5 is another way to archive beatiful UI.

webspiderteam, 2022-05-19:
logging data is simple but chart and replay functions are complex. Im using this for monitoring only. also it can usable for youtube OSB Studio.
Python is food but gives readings line by line. 

For now I'm trying to convert tı WPF. If i can do i will upload code.

kris0725pl, 2022-06-08:
exacly i have this error on aneng st207. Pairing error...when clickonlyimage i got BIG white text error

closed 2022-07-02 by webspiderteam