webspiderteam/Bluetooth-DMM-For-Windows #2: Aneng ST207 - Bluetooth pairing error

kris0725pl, 2022-05-31:
Hi, i have ST207 and app on windows working fine. But when i connect my device i got BIG TXT EROOR. Please add this device for this app :)

webspiderteam, 2022-05-31:
I can add device but I dont have this device so I need data samples. Also are you using release or source code?

kris0725pl, 2022-06-08:
I dont have any sample...Pairing error on windows 10.

webspiderteam, 2022-06-08:
You dont leed to pair these devices. The right way is clicking on the icon to connect. Big text error means program cannot decode gatt data right way. to fix that issue i have to decode your devices datas. so I need your devices gatt data. 

For obtaining gatt data you can use Bluetooth LE Explorer from windows app store. main usage is same as my app. but after connected choose 65520 and from characteristic choose 65524 then select hex mode.
First of all I need Auto mode data.

kris0725pl, 2022-06-08:
Ok tommorow i install pp nd get information

Wysłane z aplikacji Poczta dla systemu Windows 

Od: webspiderteam
Wysłano: środa, 8 czerwca 2022 23:38
Do: webspiderteam/Bluetooth-DMM-For-Windows
DW: kris0725pl; Author
Temat: Re: [webspiderteam/Bluetooth-DMM-For-Windows] Aneng ST207 - Bluetoothpairing error (Issue #2)

You dont leed to pair these devices. The right way is clicking on the icon to connect. Big text error means program cannot decode gatt data right way. to fix that issue i have to decode your devices datas. so I need your devices gatt data.
For obtaining gatt data you can use Bluetooth LE Explorer from windows app store. main usage is same as my app. but after connected choose 65520 and from characteristic choose 65524 then select hex mode.
First of all I need Auto mode data.

webspiderteam, 2022-06-11:
I don't Know which data protocol is using for your device but it could be 10byte protocol so I added it and this release can work for you. Can you test it?
https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/releases/tag/WPFAlpha

kris0725pl, 2022-06-12:
thanx! this working...but no in 100% correctly...i made video on youtube and you see what is going on :)

webspiderteam, 2022-06-12:
can u give me a link?

webspiderteam, 2022-06-12:
https://www.youtube.com/watch?v=SZgtdLPYhs0
https://www.youtube.com/watch?v=AiduT8sY7oc&t=1532s
videos from Krzysztof Nowacki. I think thats you but I coldnt see it. Both video has android app.

kris0725pl, 2022-06-12:
yup..it is me :) new video with two app on windowws :)
https://www.youtube.com/watch?v=rJkatoVu8FE

webspiderteam, 2022-06-12:
ok I saw :) some of symbols are different in your device. Readings are ok but symbols has problem. I can fix it but as I said before I need data. so I uploaded new release with log function. With your log I can fix the issue :)
hrere it is. 
https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/releases/tag/WPFAlpha_1.02

if you send me data from your log it would be easy. 
Thank for your help :)

kris0725pl, 2022-06-12:
thank you master :) i send my log and waiting for correct version :) And i have questions...it is possible add gui with android app? looks nice :)
[log.txt](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/8886223/log.txt)

webspiderteam, 2022-06-12:
> And i have questions...it is possible add gui with android app? looks nice :)
Yes, I know it looks good but I'm new in wpf also c# :) thas why I'm doing things slowly :) I wish I can do it but its a long way for now.

Analising the data right now. I think I can upload after a couple hours :)

webspiderteam, 2022-06-12:
I think I found all symbols but u have to check it if its OK.
https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/releases/tag/WPFAlpha_1.03
Please inform me maybe there can be something wrong eg. wrong symbols :)

kris0725pl, 2022-06-12:
Wooooorking great! Very good job my friend :) Do you have paypal for little donate cafe? 
My video with version 1.0.3 :)
https://www.youtube.com/watch?v=CBDvwMhPZcQ

webspiderteam, 2022-06-12:
I'm happy yo hear that it works :)Thank you:) No we dont have it in here also you dont need to do that but thank for support :)
I'll waych now :)

webspiderteam, 2022-06-12:
Oh. I forgot to add peek and inrush values to display :( added right now ;)
https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/releases/tag/WPFAlpha_1.04

kris0725pl, 2022-06-13:
Showing perfect Inrush information! Thank you master :)

webspiderteam, 2022-06-15:
New release uploaded with chart :)

kris0725pl, 2022-06-16:
Hi..what is chart? When i turn off multimeter on application is working fine :) All data gone :) On logging is available some info about turn on and off backlight?

webspiderteam, 2022-06-16:
I mean data chart (graph) like android app O though you wanted it and desing is changed a little. Connect button hidden now and almaya on top function added. 

The device is not sending data about backlight. sorry ı cant do anything about that.

closed 2022-07-02 by webspiderteam