webspiderteam/Bluetooth-DMM-For-Windows #3: False Arrow? (ANENG V05B)

ludwich66, 2022-06-30:
News from Ludwich - i have received my new BT Dongle ;-)

issue: i found a HV arrow  (ANENG V05B)

- The sign is not in real displayed
- is the end of the arrow missing?

[image]

ludwich

webspiderteam, 2022-06-30:
Greetings :) 
I'll look at it maybe inverted the value by mistake. BTW which version is this. in older version hait same?

No en of arrow is not there because its text type indicator and i couldnt find real lighting in fonts. May be I can chance it with a icon later.

webspiderteam, 2022-06-30:
yes I had to invert it before because type 1 has it inverted. but yours not. I have to chance checking it by type. may be our H.V sign place is wrong I dont know. is it same in chengbo's app?

webspiderteam, 2022-06-30:
@ludwich66 can you send me data with hv arrow and without arrow. H.V. sign can be a different place. but my device dont have it and ı cant test also.

ludwich66, 2022-06-30:
> @ludwich66 can you send me data with hv arrow and without arrow. H.V. sign can be a different place. but my device dont have it and ı cant test also. our hv bit can be battery icon!

--> arrow is always on !! ludwich

have more time after 19:00
ludwich

ludwich66, 2022-06-30:
Data V05B with hv-arrow
[log.txt](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/9021462/log.txt)

webspiderteam, 2022-06-30:
> arrow is always on !! ludwich

I mean real arrow on device not on app.  BTW battery icon is at 31.st bit on raw data for my device.

> Data V05B with hv-arrow
> [log.txt](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/9021462/log.txt)

arrrow on device or app?

ludwich66, 2022-06-30:
Full Power! ;-)
[log.txt](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/9021836/log.txt)

webspiderteam, 2022-06-30:
Thanks :) I think I handled it :) I will upload new version soon. Adding more chart functions zoom pan and mouse over data display on chart and settings page. these are almost done but Im working on add saved data on chart and same time new values can be seen same time. I'm about to finish but need may be one or two days.

webspiderteam, 2022-07-02:
@ludwich66 Just Uploaded new release :)
Only Chart data import and export has left. I'll do it soon.

ludwich66, 2022-07-02:
Now the arrow is gone - Well done!
ludwich

closed 2022-07-08 by webspiderteam