webspiderteam/Bluetooth-DMM-For-Windows #4: V1.08 Missing Parts in Installation Version

ludwich66, 2022-07-07:
Hi Webspiderteam,

i testet short both versions.

Portable Version starts normal, Installation Version produce an Error Message "there are missing parts" -- ??

I have one wish - can you add the device selection by double-clicking it :-)

Thanks ludwich

webspiderteam, 2022-07-08:
Done :) added double click and re-created the installer pack ;) But I'm not sure about installer because its Visuals Studios Default Publish module. I have no control on it.
I didnt changed release name so you can download https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/releases/tag/WPFRelease1.08 from here

webspiderteam, 2022-07-08:
Its good to hear that 😊

I'm not planning that right now but may be later because I think the import graph gives much more than CSV. A list of readings not enough to understand data I think. But it could be.

kris0725pl, 2022-07-10:
Hi master! Version 1.0.8....whoa. tommorow download and revie on youtube ;)

webspiderteam, 2022-07-10:
Adding one last thing right now. 
[image]

It can ready after a couple hours.

webspiderteam, 2022-07-10:
Done :) Now its release 1.09 )
https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/releases/tag/WPFRelease1.09

kris0725pl, 2022-07-10:
holy crow! looks amaizing!!!! Now i obtain some repositories stlink v2 on github...but i today download this version and recording video to youtube users...you are MASTER CHIEF :) Regards from Poland

webspiderteam, 2022-07-10:
Sorry for problem. Visual Studios build in Publish module sometimes gives this problem. so I changed the Installer App. Now It should work without problem.

webspiderteam, 2022-07-22:
New release available :)
https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/releases/tag/WPFRelease1.10

kris0725pl, 2022-07-22:
Ok..i got it today and test :) thx for info :) 
  Dnia 22 lipca 2022 15:34 webspiderteam   napisał(a):  
 New release available :)  
 github.com https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/releases/tag/WPFRelease1.10

webspiderteam, 2022-07-22:
Ohh it's not about testing just for informing if you interest new features 😊 of course if there is an error tell me then I can fix 😊 I'm testing it so much with my device an PC but other than my PC can have issues

closed 2022-07-23 by ludwich66

webspiderteam, 2022-07-25:
@kris0725pl uploaded new release with multi language :) 
if You wish to download 
https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/releases/tag/WPFRelease1.12

kris0725pl, 2022-07-25:
Got it! I see pl languge. But not working..in folder dont have folder pl

Wysłane z aplikacji Poczta dla systemu Windows 

Od: webspiderteam
Wysłano: poniedziałek, 25 lipca 2022 20:52
Do: webspiderteam/Bluetooth-DMM-For-Windows
DW: kris0725pl; Mention
Temat: Re: [webspiderteam/Bluetooth-DMM-For-Windows] V1.08 Missing Parts inInstallation Version (Issue #4)

@kris0725pl uploaded new release with multi language :)
if You wish to download
https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/releases/tag/WPFRelease1.12

webspiderteam, 2022-07-25:
Done :) Uploaded to https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/releases/tag/WPFRelease1.12 
same release page but added new files with Polish

kris0725pl, 2022-07-25:
Ok i got
Fix this
Button Save now is Ratowac, Change to Zapisz.

Łączyć change to Połącz 

Wysłane z aplikacji Poczta dla systemu Windows 

Od: webspiderteam
Wysłano: poniedziałek, 25 lipca 2022 21:14
Do: webspiderteam/Bluetooth-DMM-For-Windows
DW: kris0725pl; Mention
Temat: Re: [webspiderteam/Bluetooth-DMM-For-Windows] V1.08 Missing Parts inInstallation Version (Issue #4)

Done :) Uploaded to https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/releases/tag/WPFRelease1.12
same release page but added new files with Polish

webspiderteam, 2022-07-25:
[BluetoothDMM.resources.dll.zip](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/9184334/BluetoothDMM.resources.dll.zip)
Done :) put this dll to "pl" folder under app folder