webspiderteam/Bluetooth-DMM-For-Windows #5: V1.10

ludwich66, 2022-07-22:
It's on going _Thnks ;-)

Reviewcomments 1.10:

- Viewable Time of SlashScreen its a little bit to fast (Iam to slow ;-)
- Is there a link to your Github Page? - if not- Please add ABOUT / OVER 

Data processing (wishes):
- You are not a friend of csv Dataexport - what do you mean about MQTT to send life data  ?
 MQTT-Client(BTMM-Device-Software) --publish--> Broker/MQTT-Server(zb. MOSQUITTO) <--subscribe --NodeRed Dataanalysis / Database etc.
Example: (Tasmota MQTT message)
17:51:10.881 MQT: tele/Bluetooth-DMM_AA:BB:cc:dd/SENSOR = {"Time":"2022-07-22T17:51:10","BTDMMx":{"V":12.80}

Its usable for longtime measurements, Accucharging....

Devicemapping:
please made a complete list of brandnames Or a other way?,but this is irritable if you use a ZOYI ZT-300AB and your Software
meens "known as Aneng9002"

Manufacturer -> Brands

- Name: BluetoothDDM (device known as BluetoothDMM)
- Name: BluetoothDDM (device known as Aneng9002)

Thanks an have Fun

ludwich

webspiderteam, 2022-07-22:
> * Viewable Time of SlashScreen its a little bit to fast (Iam to slow ;-)

Its not changable it depends on loading time of app but I'll try

> * Is there a link to your Github Page? - if not- Please add ABOUT / OVER

No there is not yet I didn't need it but I can. 

> Data processing (wishes):
>
> * You are not a friend of csv Dataexport - what do you mean about MQTT to send life data  ?
>   MQTT-Client(BTMM-Device-Software) --publish--> Broker/MQTT-Server(zb. [...]

I have no Idea about protocol but I've searched and its about sending data to another places I think. Do we need that in your opinion? and Why?

This app can record data 1M seconds (277.77 Hours means more than 10 days) 

> Devicemapping:
> please made a complete list of brandnames Or a other way?,but this is irritable if you use a ZOYI ZT-300AB and your Software
> meens "known as Aneng9002"

Oh its not brand name or eg. its custom name that user can change for app. For example if you have 3 different device and when you want to connect from app All of them has same Name "Bluetooth DMM".  User can change their name like me "ANeng An9002" or "My clamp Meter" e.g. I cant find any brand name info yet.

Thank you for advices.

ludwich66, 2022-07-22:
> > * Viewable Time of SlashScreen its a little bit to fast (Iam to slow ;-)
> 
> Its not changable it depends on loading time of app but I'll try
>>> ;-) My PC is too fast!
> 
> > * Is there a link to your Github Page? - if not- Please add ABOUT / OVER
> 
> No there is not yet I didn't need it but I can.
Its good for Orientation
> I have no Idea about protocol but I've searched and its about sending data to another places I think. Do we need that in your opinion? and Why?
oh there are many usable things:
Make Charts in NodeRed
Feed a Database 
I do Akku Charge / Dischargeanalysis 
> 
> This app can record data 1M seconds (277.77 Hours means more than 10 days)
is there are a tool to extract or reed .dcd Files?
I mean it is only graph Data
I need really "values" to analyze
 complete after a bestrew or
 aca MQTT / Other Protocol to sample single data in a data collection
> Oh its not brand name or eg. its custom name that user can change for app. For example if you have 3 different device and when you want to connect from app All of them has same Name "Bluetooth DMM". [...]
OK
How did you handle multiple devices?
Multiple parallel Programms or multi in one Applikation?
The Original software can do rules or calculations with 1-4 Devices
> Thank you for advices.
Gladly, it's always nice when you can combine different requirements and grow

CU Tomorrow

ludwich

webspiderteam, 2022-07-22:
> Its not changable it depends on loading time of app but I'll try
> 
> > > ;-) My PC is too fast!

Done :) now its minimum 3 seconds and plus 1 second fade effect but not uploaded yet maybe tomorrow

> oh there are many usable things:
> Make Charts in NodeRed
> Feed a Database
> I do Akku Charge / Dischargeanalysis

Ok if you really need I will export csv tomorrow It can finish too

> How did you handle multiple devices?
> Multiple parallel Programms or multi in one Application?

App records every connected device to database when find a device from database that can connect app asks to connect. No limit for list. if device is not connected by any app My app asks for connect. so if you open multiple instance of app it will connect one by one. Its Multiple parallel Programms. I can start every instace by app but I dont think it really needs.

> The Original software can do rules or calculations with 1-4 Devices

I can do the same but I'm not sure this really needs

CU Too :)

webspiderteam, 2022-07-23:
- [x] SplashScreen timeout Done!
- [x] About Dialog Done!
- [x] Export CSV format Done!
- [x] Upload new release Done!

ludwich66, 2022-07-23:
Example Chart to introduce MQTT
Example Use BT-DMM to Analyse  Data from charge / discharge / check an Akku and cut or feed Chargevoltige via an Relais Actor.

[Chart_MQTT.odp](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/9174150/Chart_MQTT.odp)

You are Faster with you Implementing, I check it in the Evening.

Thnx before 
Ludwich

webspiderteam, 2022-07-23:
If I understand true you need realtime data transport to another app or database. Right?

ludwich66, 2022-07-23:
Yes, direct data control is possible.
The data rate is another issue. With the DMM, 3 values can be generated per/sec (highest rate). However, it is certainly sufficient to smooth values, e.g. 3 values averaged / sec (no experience how high the data rate can be. NiMh batteries have a minimal voltage drop when fully charged (-Delta V). It would be nice if you could evaluate it - it's a project to gain experience.
These are my "needs" - what do you practically do with them?
;-)

webspiderteam, 2022-07-23:
Data rate is simple I think because no need all values for creating chart or what ever. one data per second is enough. value cant change a lot in a second maybe 1 mV change can be posible thats all.  

The main issue is I have no experience with Mqtt and its deep waters for me right now. First of all I need a server for Mqtt that receives data and I should know the protocol and data structure. if there is a server i can make a client. Because I have to test it.

closed 2022-07-23 by ludwich66