webspiderteam/Bluetooth-DMM-For-Windows #6: V1.11

ludwich66, 2022-07-23:
First Test:

bad news -> 4 crashes
- Communication Window "don't ask again" ignored once
-  open with activated Chart don't word now (change ignored, always off)
- Versionslabel in Main and Aboutwindow is  "1.10" eg "1.11"
- Application closed after export csv file ? 3 exits without export a file
- Exportextension: preselected is first dcd, second csv. csv is in my opinion more common than uncommon dcd-file. is it possible to change preselect

+++ for "About Window"
!!! Please add Timestamp to csv File (DMM is not precise 3 / Sec)
later:
Add a little help, - Info Export formats, Info 1M max Samples , Savetime , How to deactivate "Powersafe" on DMM

more after my family meeting ;-)

graet way to grow!!

webspiderteam, 2022-07-23:
_* Communication Window "don't ask again" ignored once_
    Dont ask mean connect directly without ask. but fixed disabled ignore button when dont ask checked :)
_* open with activated Chart don't word now (change ignored, always off)_ 
    I tried but does not give a problem. maybe somethink else effects it i will try to find. Maybe settings not saved. Can u try again?
_* Versionslabel in Main and Aboutwindow is  "1.10" eg "1.11"_
   Fixed :) I was changed version but it was a wrong place :)
 _* Application closed after export csv file ? 3 exits without export a file_
   I tried more than 10 times gives no error. is there a special char in path that you want to save?
 _* Exportextension: preselected is first dcd, second csv. csv is in my opinion more common than uncommon dcd-file. is it possible to change preselect_
   Its posible but app does not import from csv. when user try to import they cant. but i can put a option in settings to made it default.

ludwich66, 2022-07-23:
* open with activated Chart don't word now (change ignored, always off)
I tried but does not give a problem. maybe somethink else effects it i will try to find. Maybe settings not saved. Can u try again?
-> Installed Version dosen't save
-> Portable Version saves after Changing Folder Without " . " in Pathname
    Its Standardname Downloaded Githubfile with Extract
[image]

* Application closed after export csv file ? 3 exits without export a file
always the Same, i used now Path of Portable Program, Filename "exp1.csv"
4 Programstart - 1 save File, 3 Times Programm closed without saving

Lets try Tomorrow - we find the Bug :-)

ludwich

webspiderteam, 2022-07-24:
[installer.zip](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/9175339/installer.zip)
[portable.zip](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/9175347/portable.zip)

try these. it has debug.txt in app folder and will give us whats the problem or where.

ludwich66, 2022-07-24:
thanks for debug version

First event: Save CSV / Save to Same Folder
second try - app closed

```
start export
dialog with resultTrue Filename: C:\Users\reteq\Desktop\portable\export_port_1.csv
extention .csv
export
start exportcsv
chart data 40
create text41
first text
plottable list done
added last plottable
error catch Error: System.IndexOutOfRangeException: Der Index war außerhalb des Arraybereichs.
   bei BluetoothDMM.MainWindow.Chart_Export_Click(Object sender, RoutedEventArgs e) in C:\Users\Engin\Source\Repos\Bluetooth-DMM-For-Windows\BluetoothDMM WPF\BluetoothDMM\MainWindow.xaml.cs:Zeile 885.
```

webspiderteam, 2022-07-24:
ok we are getting somewhere.
[Release.zip](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/9175632/Release.zip)
this should give us exact point of where the error

ludwich66, 2022-07-24:
I short to lunch

But feed you first :

start export
dialog with resultTrue Filename: C:\Users\reteq\Desktop\portable\export1.csv
extention .csv
export
start exportcsv
chart data 87
create text88
first text
plottable list done
added last plottable
item 0ct len: 88 keyvallen: 9
pairs 0create text:System.String[]ct len: 88 keyvallen: 9
item added 0
item 1ct len: 88 keyvallen: 9
pairs 0create text:System.String[]ct len: 88 keyvallen: 9
item added 1
item 2ct len: 88 keyvallen: 9
pairs 0create text:System.String[]ct len: 88 keyvallen: 9
item added 2
item 3ct len: 88 keyvallen: 9
pairs 0create text:System.String[]ct len: 88 keyvallen: 9
item added 3
item 4ct len: 88 keyvallen: 9
pairs 0create text:System.String[]ct len: 88 keyvallen: 9
item added 4
item 5ct len: 88 keyvallen: 9
pairs 0create text:System.String[]ct len: 88 keyvallen: 9
item added 5
item 6ct len: 88 keyvallen: 9
pairs 0create text:System.String[]ct len: 88 keyvallen: 9
item added 6
item 7ct len: 88 keyvallen: 9
pairs 0create text:System.String[]ct len: 88 keyvallen: 9
item added 7
item 8ct len: 88 keyvallen: 9
pairs 0create text:System.String[]ct len: 88 keyvallen: 9
item added 8
item 9ct len: 88 keyvallen: 9
pairs 0create text:System.String[]ct len: 88 keyvallen: 9
item added 9
item 10ct len: 88 keyvallen: 9
pairs 0create text:System.String[]ct len: 88 keyvallen: 9
item added 10
item 11ct len: 88 keyvallen: 9
pairs 0create text:System.String[]ct len: 88 keyvallen: 9
item added 11
item 12ct len: 88 keyvallen: 9
pairs 0create text:System.String[]ct len: 88 keyvallen: 9
item added 12
item 13ct len: 88 keyvallen: 9
pairs 0create text:System.String[]ct len: 88 keyvallen: 9
item added 13
item 14ct len: 88 keyvallen: 9
pairs 0create text:System.String[]ct len: 88 keyvallen: 9
item added 14
item 15ct len: 88 keyvallen: 9
pairs 0create text:System.String[]ct len: 88 keyvallen: 9
item added 15
item 16ct len: 88 keyvallen: 9
pairs 0create text:System.String[]ct len: 88 keyvallen: 9
item added 16
item 17ct len: 88 keyvallen: 9
pairs 0create text:System.String[]ct len: 88 keyvallen: 9
item added 17
item 18ct len: 88 keyvallen: 9
pairs 0create text:System.String[]ct len: 88 keyvallen: 9
item added 18
item 19ct len: 88 keyvallen: 9
pairs 0create text:System.String[]ct len: 88 keyvallen: 9
item added 19
item 20ct len: 88 keyvallen: 9
pairs 0create text:System.String[]ct len: 88 keyvallen: 9
item added 20
item 21ct len: 88 keyvallen: 9
pairs 0create text:System.String[]ct len: 88 keyvallen: 9
item added 21
item 22ct len: 88 keyvallen: 9
pairs 0create text:System.String[]ct len: 88 keyvallen: 9
item added 22
item 23ct len: 88 keyvallen: 9
pairs 0create text:System.String[]ct len: 88 keyvallen: 9
item added 23
item 24ct len: 88 keyvallen: 9
pairs 0create text:System.String[]ct len: 88 keyvallen: 9
item added 24
item 25ct len: 88 keyvallen: 9
pairs 0create text:System.String[]ct len: 88 keyvallen: 9
item added 25
item 26ct len: 88 keyvallen: 9
pairs 0create text:System.String[]ct len: 88 keyvallen: 9
item added 26
item 27ct len: 88 keyvallen: 9
pairs 1create text:System.String[]ct len: 88 keyvallen: 9
item added 27
item 28ct len: 88 keyvallen: 9
pairs 2create text:System.String[]ct len: 88 keyvallen: 9
item added 28
error catch Error: System.IndexOutOfRangeException: Der Index war außerhalb des Arraybereichs.
   bei BluetoothDMM.MainWindow.Chart_Export_Click(Object sender, RoutedEventArgs e) in C:\Users\Engin\Source\Repos\Bluetooth-DMM-For-Windows\BluetoothDMM WPF\BluetoothDMM\MainWindow.xaml.cs:Zeile 885.

webspiderteam, 2022-07-24:
[Release.zip](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/9175679/Release.zip)
ok this should fix the issue.

ludwich66, 2022-07-24:
+++
+ great saving of dcd and csv worked without problems

---
- hook  by "Data Chart" is lost every start
- Settings "Chart ON on startup" is activated
- File: BluetoothDMM.exe.config
<userSettings>
    <BluetoothDMM.Properties.Settings>
      <setting name="Ontop" serializeAs="String">
        <value>False</value>
      </setting>
      <setting name="ChartOn" serializeAs="String">
        <value>**False**</value>
      </setting>
      <setting name="WindowPosition" serializeAs="String">
- 
```
CU ludwich

webspiderteam, 2022-07-24:
in menu Data chart has thick? even if its not shows up?
File: BluetoothDMM.exe.config has default config not saves current value in it.

webspiderteam, 2022-07-24:
in C:\Users\reteq\AppData\Local  
delete or rename WebSpiderTeam folder thats where they stored. you can also check for values in user.config

ludwich66, 2022-07-24:
Please see Video "Loose Chart On"
https://youtube.com/shorts/QxRmx5YUQzo

webspiderteam, 2022-07-24:
can you try it without save position and size

ludwich66, 2022-07-24:
Yes

webspiderteam, 2022-07-24:
Ok i think i found the problem. When save position and size enabled app checks the size of window and if its not enough to show chart it increases the size but when i made analog meter i forgot to change checked size values but now its fixed.

> there are holes in the export column 2 "DC" and "V" is this an Bug or a feature? ;-)

its a bug but not compeletely. in AUTO mode there is no ac/dc and symbol on screen but chart was reading the data. now I changed it. now if it says AUTO (not Auto Range) and NCV mode (these 2 has no ac/dc and symbol) chart data will not continue until reads a real value.
[Release.zip](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/9175965/Release.zip)
check this. this should work as expected :)

ludwich66, 2022-07-24:
I meen V1.12 was born ;-)

but one little optical point, is it possible to change contrast of activ Bargraph to Inactive Part?
Used Colors differs not much. ( on my Display with my Eyes ;-)

What's the next step?

Ludwich

webspiderteam, 2022-07-24:
can you explain a bit more (my english is not enough sometimes :))

ludwich66, 2022-07-24:
Handdrawn ;-)
I meen a higher contrast between scale and "Pointer" ( i dont know wright word in English)
I hope my example helps
[image]

1.11 with Bugfix its Ok :-) - it was the message - it works properly!
normaly 1.11.1 was the Bugfix of 1.11, other users should be able to differentiate the versions.

Good work

webspiderteam, 2022-07-24:
ok got it :) My scren is in color blindless mode bcuz I'm :)
try this ;)
[Release.zip](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/9176119/Release.zip)

webspiderteam, 2022-07-24:
Uploaded fixed releases

ludwich66, 2022-07-24:
+++ The yellow pointer is the Better way.
Look good on my LCD Screens.
Get in touch if it goes on, I'll be happy to check the functions!

Ludwich

webspiderteam, 2022-07-24:
Thanks for all :) I will inform to you ;)
yes lets close :)

ludwich66, 2022-07-24:
the yellow pointer is in my opinion better

ludwich66, 2022-07-24:
a new little issue.
there is a running measurement ,  i open the setting menu and klick save - in the main window number-display and Graph works well -- Bargraf shows nothin left side.
its not an showstoper for this Version...

webspiderteam, 2022-07-24:
What was your settings?

webspiderteam, 2022-07-24:
Showstopper? İf you mean when mouse is over graph it stops scrolling. Still has same feature

ludwich66, 2022-07-24:
I have analog an graph activate, then i go in the setting menu and click save Button.
analog Bargraph shows after save "0" left side, Bargraph do not refresh (show analog off / on)
like a Video?
;-)

webspiderteam, 2022-07-24:
can you show with screenshot?

ludwich66, 2022-07-24:
Video https://youtu.be/lLdtMhmQThM

webspiderteam, 2022-07-24:
ok got it :) fixed :)
[BluetoothDMM.zip](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/9176344/BluetoothDMM.zip)
this is just exe extract and copy exe to your app folder

ludwich66, 2022-07-24:
Yes looks good - Should I keep looking or is that enough for today ;-)

webspiderteam, 2022-07-24:
Its your choise :) if you dont have anything to do and if you are boring you can continue :) else you can continue any time you want :) 
Thanks for helping. Probably I cant find these issues without you 🤝

ludwich66, 2022-07-24:
;-) I Stop for Today - Familytime

Tomorrow I'm collecting information about MQTT and Windows (VS) ;-)

closed 2022-07-24 by ludwich66

webspiderteam, 2022-07-25:
It's test time :) new release is coming :)
[Portable.zip](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/9183699/Portable.zip)
If you wanna look :) not uploading yet ;)