webspiderteam/Bluetooth-DMM-For-Windows #7: Review 1.12

ludwich66, 2022-07-26:
Hi Engin ( ok for you?)

where can i best change translation files. is it saved as .dll?

ludwich

webspiderteam, 2022-07-26:
I already added Polish and sended :) 

you need german translation I think
[de.csv](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/9189095/de.csv)
[en.csv](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/9189097/en.csv)
here in csv format you can edit and send me back then I can compile as dll

ludwich66, 2022-07-26:
Hi Engin ,
first "Almanca"

"not connected" is first not translated - changes after established connection

Translation first try - check space or cuted literals in second step
[de.csv](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/9189419/de.csv)

+++

Ludwich

webspiderteam, 2022-07-26:
[de.zip](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/9189985/de.zip)
copy files inside zip and paste to app folder

webspiderteam, 2022-07-26:
> "not connected" is first not translated - changes after established connection

I know I fixed it but not published yet

> Translation first try - check space or cuted literals in second step

I saw it in device management and connection popup 
I fixed connection poppup and trying find a way to fix in device manegement

webspiderteam, 2022-07-26:
> * ist there a option to read "localisation" from windows and use this info as preselection?

already done :) but I dont know published release has it or not

webspiderteam, 2022-07-26:
[Release.zip](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/9190282/Release.zip)
posibly this has all you want :)

ludwich66, 2022-07-26:
Is Version 1.12 on Relaespage actually now?
I sent messages to Youtube Users with the wish to help Translate an check.
Please actualize it soon. 

Ludwich

webspiderteam, 2022-07-26:
not this one but close enough I can upload a new release too

ludwich66, 2022-07-26:
+++ Please 👍 +++

webspiderteam, 2022-07-26:
do you think this is 1.13? or just 1.12a?

webspiderteam, 2022-07-26:
1.13 Released :)

closed 2022-07-26 by ludwich66