webspiderteam/Bluetooth-DMM-For-Windows #13: file export mode

ks00x, 2022-08-02:
I think it will be better to have 'csv' as the default for exporting the data. Maybe add a column for the full values in exponential format (without the u,m,k,M... prefixes).

webspiderteam, 2022-08-02:
I will add an option to settings to make it default output.

> Maybe add a column for the full values in exponential format (without the u,m,k,M... prefixes).

Can you give an example

ks00x, 2022-08-02:
> Can you give an example
now:
49,DC,-0.23,mV
new column (in V in this case):
49,DC,-0.23,mV,-2.3E-4

webspiderteam, 2022-08-02:
I got it. You mean 1.23mV = 0.00123 V and 1Mohm as 1000000ohm. Right?

ks00x, 2022-08-02:
> I got it. You mean 1.23mV = 0.00123 V and 1Mohm as 1000000ohm. Right?

yes, makes sense on an autorange meter to me.

ludwich66, 2022-08-02:
Das Thema mit CSV ist die gleiche Baustelle wie im MQTT-Stream. Problem erkannt - Problem gesperrt!
Problem erkannt - Problem gebannt!
;-) ludwich

webspiderteam, 2022-08-02:
done :) new release uploaded

ludwich66, 2022-08-02:
High time to keep an eye on you (nearly tomorrow)
Thank you for the time invested, my wishes have not turned out to be entirely impractical ;-)

closed 2022-08-05 by webspiderteam