webspiderteam/Bluetooth-DMM-For-Windows #14: Review 115

ludwich66, 2022-08-02:
Refresh MQTT string
1: Set to °F -->> Value 77
2: Change to Diodemode
Issue: ValueF is filled with Value from last measurement
{"Time":"02.08.2022 22:29:19","Device":"DMM_V05B","ValueS":" .0L ","ValueF":77,"Range":"V","Current":"","Symbols":"BUZZ,DIODE"}

webspiderteam, 2022-08-02:
thats what I said. tomorrow I will lookup

ludwich66, 2022-08-02:
good night today

ludwich66, 2022-08-03:
hi engin,
which iconsets do you use?
Please tell me best size and format for icons

ludwich

webspiderteam, 2022-08-03:
> Refresh MQTT string

I think I should change ValueF as null when not numerical. thats the best way to handle.

for which icons? App icon? Graph popup? Menu? Flags? Device Selector? Main Display?

webspiderteam, 2022-08-03:
most of them is svg format so resizeable only images that I used is here
https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/tree/master/BluetoothDMM%20WPF/BluetoothDMM/Assets

ludwich66, 2022-08-03:
Draft icons for BT-DMM
interested right?

[BT-DMM_ICON-SET.zip](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/9251215/BT-DMM_ICON-SET.zip)

ludwich66, 2022-08-03:
View on Top2
[image]

webspiderteam, 2022-08-03:
[image]
what do you think?

ludwich66, 2022-08-03:
more wishes?
[image]

ludwich66, 2022-08-03:
last-complete-set

[BT-DMM-ICONSET2.zip](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/9252050/BT-DMM-ICONSET2.zip)

ludwich66, 2022-08-03:
> [image]
> what do you think?

Yes looks good ;-)

Is it possible to move your Setting-Wheel from right to the left menu entries 
AND
bring language setting (or button to open subpage ) in first order of basic MENU?

... and you need a Copyright Legal Info

But that's Eye- and Ergonomic-Candys - Base functions must have Priority.

You are a great developer !

Did you know my nikname at my job?
"merciless ruler"

;-) ludwich

  | occupation |  
-- | -- | --

<br class="Apple-interchange-newline" style="caret-color: rgb(0, 0, 0); color: rgb(0, 0, 0); font-style: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-size-adjust: auto; -webkit-text-stroke-width: 0px; text-decoration: none;">

ludwich

closed 2022-08-05 by webspiderteam