webspiderteam/Bluetooth-DMM-For-Windows #29: Switching metering mode from bluetooth?

PBrunot, 2022-11-15:
Hello,
Is it possible to switch from voltage measurement to resistor measurement via bluetooth ? The DMM app was able to switch between temperature and capacitance testing.

webspiderteam, 2022-11-15:
Yes it's possible but we couldn't find any clue of how app does it. I saw there is another characteristics possibly for that reason. Also my 9002 hasn't got this ability so I can't test it. I wish I could do but I will try to find it again

PBrunot, 2022-11-15:
Ill give it a try too following this
https://stackoverflow.com/questions/23877761/sniffing-logging-your-own-android-bluetooth-traffic

webspiderteam, 2022-11-15:
> Ill give it a try too following this https://stackoverflow.com/questions/23877761/sniffing-logging-your-own-android-bluetooth-traffic

I don't know if it's work but if you can find anything about it tell me please

ludwich66, 2022-11-15:
> Ill give it a try too following this https://stackoverflow.com/questions/23877761/sniffing-logging-your-own-android-bluetooth-traffic Le mar. 15 nov. 2022, 08:12, webspiderteam ***@***.***> a écrit :
> […](#)

I've got the little multimeter here - if we have more information on how to control the DMM I'm happy to help.
Ludwich

PBrunot, 2022-11-15:
I did a capture on Android phone with DMM app connected to V05B bluetooth multimeter.
I repeatly clicked on Temperature Capacitance Auto button from the apps which toggle a different mode in the meter.
[Then following this procedure : ](https://medium.com/@charlie.d.anderson/how-to-get-the-bluetooth-host-controller-interface-logs-from-a-modern-android-phone-d23bde00b9fa) I captured the attached trace.

[btsnoop_hci.log](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/10015406/btsnoop_hci.log)

I did not yet install Wireshark to parse it, but it should contains the writes on the BT characteristics issued by the original app.

webspiderteam, 2022-11-15:
Looks like it sends 10 byte data to dmm for 0xfff4 characteristic. there is 2 bytes changing
[image]

webspiderteam, 2022-11-15:
@ludwich66 can you test these values and check for responses on multimeter? we need to know which data makes what difference.

ludwich66, 2022-11-15:
> @ludwich66 can you test these values and check for responses on multimeter? we need to know which data makes what difference.

Do you have a idea how i do a Test this?
L

PBrunot, 2022-11-15:
> > @ludwich66 can you test these values and check for responses on multimeter? we need to know which data makes what difference.
> 
> Do you have a idea how i do a Test this? L

Download "NRF connect" on your smartphone, connect to DMM meter, write the characteristic with byte array as per example below. Test with the first values. You can even save those byte arrays in the app.

webspiderteam, 2022-11-15:
> Download "NRF connect" on your smartphone, connect to DMM meter, write the characteristic with byte array as per example below. Test with the first values. You can even save those byte arrays in the app.

that can work. also my old app is based on Bluetooth Le Explorer on windows app store and it has same feauture too

ludwich66, 2022-11-15:
[image]

last Part for today ;.-)

1: AUTO = eaec70eda2c132716499
2: °C = eaec70e3a2c13271649b
3: °F  = eaec70e5a2c132716481
4: Capa =
5: Buz/Diode = eaec70e4a2c132716486
6: NCV
7: Hz = eaec70e6a2c132716484
8: ZERO
9: mA / A (only with switched Probe to left Place)

L

webspiderteam, 2022-11-16:
> [image]
> 
> last Part for today ;.-)
> 
> 1: AUTO = eaec70eda2c132716499 2: °C = eaec70e3a2c13271649b 3: °F = eaec70e5a2c132716481 4: Capa = 5: Buz/Diode = eaec70e4a2c132716486 6: NCV 7: Hz = eaec70e6a2c132716484 8: ZERO 9: mA / A (only with switched Probe to left Place

Great 😃 
[image]

What about this Page

webspiderteam, 2022-11-16:
> Capa = WRONG eaec8ee5a2c1327165bf

@ludwich66 there in no record for "eaec8ee5a2c1327165bf"  all of them is "ea ec 70 (e6) a2 c1 32 71 64 (84)"

ludwich66, 2022-11-16:
> > Capa = WRONG eaec8ee5a2c1327165bf
> 
> @ludwich66 there in no record for "eaec8ee5a2c1327165bf" all of them is "ea ec 70 (e6) a2 c1 32 71 64 (84)"
 
Morning Engin,
I tried the Android tool "LightBlue". It delivers this key after the button is pressed in the Android BTDMM app.
It doesn't fit into the pattern but is different for each key. I don't know if there's a key on it.
I haven't been able to read "ea ec 70 ..." yet...
currently
L

PBrunot, 2022-11-16:
I will test methodically tonight and send the results.

PBrunot, 2022-11-16:
I played with V05B and I think the confiusing logic in the bluetooth application button actually mimicks buttons and the multimeter behaviour itself.

It is a "smart multimeter" in the sense that it decides what it can measure.

There are two main modes:
A) test lead in V /Ohms/Diode -> Current measurement blocked 
B)  test lead in A mA hole -> all other measurements blocked

Bluetooth interface cannot force switches from A) to B)

In mode A) you can switch with "SEL/NCL" button from AUTO (A.0) to continuity testing (A.1) to capacitance testing (A.2) to frequency (A.3) to temperature °C (A.4) to temperature °F (A.5). Non contact leads finding (A.6) can be activated by long press on NCL button.

So the bluetooth interface can only control transitions betwen A.0.x / A.1 / A.2 / A.3 / A.4 / A.5 / A.6 like the physical button.

When the meter is in AUTO state (A.0) it will choose between V measurement (A.0.1) and Ohms measurement (A.0.2) automatically and you cannot force it in anyways. I guess it senses for voltage first, then choose DC or AC, and fall back to ohm meter.

Bluetooth interface can also force HOLD MODE, I did not figure out what ZERO mode means.

In my scenario I wanted to force V / Ohms / Ampls reading from an external apps so this makes the V05B not really useful and the complex interaction by the meter makes, in my opinion, the whole idea of controlling the mode from another app quite confusing and useless.

webspiderteam, 2022-11-16:
İf it blocks amperage there is nothing to do with it. We can only use available actions

ludwich66, 2022-11-16:
Ciao Pasquale,
anche se la soft selection non ti è utile - ti chiedo supporto nella ricerca delle altre chiavi.
Ho aggiunto al mio post su di esso. Cosa puoi farne è una domanda a cui trovo sempre una risposta ;-)
Grazie
>GoogleTranslate>

ludwich66, 2022-11-16:
Technical Blocks, tomorrow i will check if it is possible to use three Probes and switch between all modes.
I mean the logic is only programmed (Android&IphoneApp) but, codes changes Modes in DMM directly.
So you can Change if it is correct wired.

One Usable Projekt for the Cold Future is a "blackout-control", Switch between Volt and Hertz. ;-)

L

PBrunot, 2022-11-16:
Ich werde dir helfen ... aber Morgen jetzt ist es zu spaet fuer mich.

PBrunot, 2022-11-17:
Hi,

I did the tests, this are my finding. Button click sequence was recorded, then I manually matched the wireshark trace to the packet. Not all behaviours can be explained.

Trace captured:
[btsnoop_hci.log](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/10036123/btsnoop_hci.log)
You can filter by destination ShenZen -> Sent Write command to see the sequence of BT writes.

Sequence and packets:

| Button click | BEEP | Packet               |
|--------------|------|----------------------|
| AUTO         | yes  | eaec70eda2c132716499 |
| NCV          | yes  | eaec70e7a2c132716487 |
| AUTO         | yes  | eaec70eda2c132716499 |
| NCV          | yes  | eaec70e7a2c132716487 |
| AUTO         | yes  | eaec70eda2c132716499 |
| NCV          | yes  | eaec70e7a2c132716487 |
| AUTO         | yes  | eaec70eda2c132716499 |
| NCV          | yes  | eaec70e7a2c132716487 |
| °C           | yes  | eaec70e3a2c13271649b |
| °F           | yes  | eaec70e2a2c132716498 |
| °C           | yes  | eaec70e3a2c13271649b |
| °F           | yes  | eaec70e2a2c132716498 |
| °C           | yes  | eaec70e3a2c13271649b |
| °F           | yes  | eaec70e2a2c132716498 |
| AUTO         | yes  | eaec70eda2c132716499 |
| CAP          | yes  | eaec70e5a2c132716481 |
| AUTO         | yes  | eaec70eda2c132716499 |
| CAP          | yes  | eaec70e5a2c132716481 |
| Hz           | yes  | eaec70e6a2c132716484 |
| AUTO         | yes  | eaec70eda2c132716499 |
| Hz           | yes  | eaec70e6a2c132716484 |
| AUTO         | yes  | eaec70eda2c132716499 |
| DIODE        | yes  | eaec70e4a2c132716486 |
| AUTO         | yes  | eaec70eda2c132716499 |
| DIODE        | yes  | eaec70e4a2c132716486 |
| AUTO         | yes  | eaec70eda2c132716499 |
| DIODE        | yes  | eaec70e4a2c132716486 |
| NCV          | yes  | eaec70e7a2c132716487 |
| AUTO         | yes  | eaec70eda2c132716499 |
| ZERO         | no   | no packet            |
| AUTO         | yes  | no packet            |
| ZERO         | no  | no packet            |
| HOLD         | no  | no packet            |
| MAX/MIN      | no  | no packet            |
| CAP          | yes  | eaec70e5a2c132716481 |
| MAX/MIN      |no  | eaec7084a2c1327164e6 | ?
| CAP          | yes  | eaec70e5a2c132716481 |
| HOLD         | no  | eaec7091a2c132716495 | ?
| HOLD         | no  | no packet            |
| V            | no  | no packet            |
| Ohms         | no  | no packet            |
| Ohms         | no  | no packet            |
| NCV          | yes  | eaec70e7a2c132716487 |
| Ohms         |no  | eaec70eba2c132716493 | ?
| MAX/MIN      | no  | no packet            |
| HOLD         | no  | no packet            |
| °C           | yes  | eaec70e3a2c13271649b |
| HOLD         | no  | no packet            |
| HOLD         | no  | no packet            |
| MAX/MIN      | no | eaec7084a2c1327164e6 |
| mVHz         | no  | no packet            |
| Ohms         | no  | no packet            |
| Unknown      |   no   | eaec7093a2c1327164eb |

ludwich66, 2022-11-17:
Oh Great,

i test it tomorrow - Its to late now ;-)

THNX
Ludwich

webspiderteam, 2022-11-18:
Now it's time to start coding I think. First of all I have to add sending characteristic function. Then I will add buttons and data's that will send.

webspiderteam, 2022-11-19:
@ludwich66 can you test this. 
[Debug.zip](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/10046747/Debug.zip)

you will see a box to send data at bottom of main window. I need to be sure if its sending data to multimeter.

ludwich66, 2022-11-19:
> @ludwich66 can you test this. [Debug.zip](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/10046747/Debug.zip)
> 
> you will see a box to send data at bottom of main window. I need to be sure if its sending data to multimeter.

After OK 
Errorbox " False Characteristic does not support write"

Working Example nrfConnect paired with v05b:
[image]

webspiderteam, 2022-11-19:
Ok. I will work on it

webspiderteam, 2022-11-19:
[Debug.zip](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/10047414/Debug.zip)

I think this will work :)

webspiderteam, 2022-11-19:
> True Success!
> 
> DMM switched to °C
> 
> ✌️
Hurray 🥇 :)

> Do you have seen zipped button pictures?

yes I saw them but Im not sure about using them. Because they are not svg format and as you said C deg and F deg has separated codes. If there is posibilities to use without pressing twice I want to do it.

webspiderteam, 2022-11-19:
Can you find anthing about HOLD function?

ludwich66, 2022-11-19:
I look for SVG Button alternative

webspiderteam, 2022-11-19:
Now its time to test with predefined buttons.
[Debug.zip](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/10047604/Debug.zip)

"->|-" is for diode mode I will change it soon

ludwich66, 2022-11-19:
Great, i made a "same" simple svgIconset ;-)

from left to Diode works - Please check AUTO, don't work and crashes App.

+++

webspiderteam, 2022-11-19:
> from left to Diode works - Please check AUTO, don't work and crashes App.

ohh sorry I forgot to seperate bytes fixed now. 
[Debug.zip](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/10047811/Debug.zip)

> Great, i made a "same" simple svgIconset ;-)

send them to me then  :)

ludwich66, 2022-11-19:
hi engin,

now auto works well !!

Order as on V05B DMM?
Auto, (V->Ω->beeper), Diode, Cap, Hz, °C, °F -> turn

HOLD /  ZERO on RubberButton

Other Codes not on Mini-DMM
MIn-Max, Ω, mV/HZ

L

webspiderteam, 2022-11-19:
[image]

arrranged the as official app

ludwich66, 2022-11-19:
Next Step Tomorrow,
CU L

ludwich66, 2022-11-20:
My First Try of Icons do you find in Folder "Extras"
L

webspiderteam, 2022-11-20:
Last test version 
[Debug.zip](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/10049418/Debug.zip)

> My First Try of Icons do you find in Folder "Extras"

I think I have done it. But thanks for helping

ludwich66, 2022-11-20:
Switching works well.
Your Iconset Wins !!

Simgeler biraz daha küçük olsaydı, ekran öğeleriyle örtüşmezlerdi. Uzak öğe etkinken hamburgerin yüksekliği hakkında? Ardından, diğer DMM'nin diğer sembolleri de uygun olabilir. Yeterli alan olup olmadığını kontrol edin.
saygılar
Gruße L

[image]

webspiderteam, 2022-11-20:
> If the icons were a bit smaller so they don't overlap the display elements

I thought it but when it's smaller some of them looks weird and some of them are not readable. That's why I made them little bigger. Also they will be hidden if mode button clicked again. But i will made mode button resized as other symbols. Because when display size is different than default it overlaps. It should look weird when other buttons is visible but I have no choice I think

webspiderteam, 2022-11-20:
[image]
for desk type multimeter

[Portable.zip](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/10049879/Portable.zip)
if this is OK, I will publish this one.

ludwich66, 2022-11-20:
Hi Pascal,

Unfortunately I can't do it myself.
Can you put some more time into those station wagons.
The previous software looks very good ;-)

HOLD
CAP --> ZERO

Did you check the latest DEBUG version?

Ludwich

webspiderteam, 2022-11-20:
> The previous software looks very good ;-)

then I'm publishing now

ludwich66, 2022-11-20:
👍

PBrunot, 2022-11-20:
Sure i will do tonight!

webspiderteam, 2022-11-20:
Hey guys :) open your dmm app :) you can update now :)

ludwich66, 2022-11-20:
I send a Mail to [Krzysztof Nowacki](https://www.youtube.com/@Kris0725PL) he use ClampDMM Aneng207.
L

ludwich66, 2022-11-20:
Thanks for the further development!

+++

webspiderteam, 2022-11-20:
> Thanks for the further development!
> 
> +++

you re welcome :)

webspiderteam, 2023-01-21:
@PBrunot All remote buttons are compleleted right now :)

PBrunot, 2023-01-21:
Well done guys. I've switched to other meters because i needed more control
of the active mode.

webspiderteam, 2023-01-21:
Thank you 😊 maybe by time you need that too ☺️ thank for everything 🙏

closed 2023-01-21 by webspiderteam