webspiderteam/Bluetooth-DMM-For-Windows #33: ST207 Missing Display Representation

ludwich66, 2022-11-23:
Displayelement is missing:
REL
PEAKHOLD

Video from Krzysztof
Min 25:40 
https://www.youtube.com/watch?v=Wwh6yD25Z7Q

webspiderteam, 2022-11-23:
Peek Hold is working. Peek translated as zerkac as seen on display and hold is showing as red numbers. But yes rel is missing because ı have no data about it. I even don't know it has rel function. I never seen it on other videos. İf you know or your device has it i can add. İt's only shows up in capacitance mode

ludwich66, 2022-11-23:
Oh, I must have misunderstood. Thank you for your prompt reply.

?? Where can I find the RAW data when the installation is complete?

L

webspiderteam, 2022-11-23:
Raw data is still under app directory with filename date-time of record dot log but not sure about raw data function is enabled by default. You have to check it in settings

closed 2023-01-17 by webspiderteam