webspiderteam/Bluetooth-DMM-For-Windows #36: ZOYI ZT-5566SE

kris0725pl, 2023-01-15:
Hi masters :) On my youtube channel i got answer about this bluetooth multimeter.

It is possible add this for work?

On aliexpress :
472,03 zł  50%OFF | ZOYI ZT-5566SE profesjonalny multimetr cyfrowy wysokiej precyzji 19999 liczy akumulator typ pulpitu Mulitimetro z Bluetooth
https://a.aliexpress.com/_EQmmVpb

Thx for all hard job!

webspiderteam, 2023-01-15:
We already added it Kris but we couldn't test it. Because we don't have any data. But teoricaly it should work. We get protocols from original app and need tested. İf you have device try to use it.

webspiderteam, 2023-01-15:
Can you send my email address to him? YouTube not allowed to me send neither my email nor link.

kris0725pl, 2023-01-15:
i reply on youtube to this topic...waiting for response :) If not...delete this topic.

webspiderteam, 2023-01-15:
Probably he hasn't got GitHub account but he can see mail address:) btw thanks for your help:)

closed 2023-01-21 by webspiderteam