webspiderteam/Bluetooth-DMM-For-Windows #46: MQTT flag to allow values in own topic

analystcmyk, 2023-10-26:
Currently topic BT_DMM/Values outputs a JSON.

Is it possible to output these values in their own topic? 

eg in MQTT settings :  

> ☑ _output each value to own topic_ 

That would mean you would see
```
BT_DMM/Values/ValueS: 4.104
BT_DMM/Values/Range: V

```

This would facilitate integrations that cannot parse JSON easily.

webspiderteam, 2023-11-10:
It could be done but every platform that I know has json decode function. but if you need help to parse I can help. 
BTW Sorry for late answer. I had health issues.

webspiderteam, 2023-11-10:
[Portable.zip](https://github.com/webspiderteam/Bluetooth-DMM-For-Windows/files/13321861/Portable.zip)

Here it is. I made a checkbox in mqtt settings you can use as

> That would mean you would see
> 
> ```
> BT_DMM/Values/ValueS: 4.104
> BT_DMM/Values/Range: V
> ```
> 
> This would facilitate integrations that cannot parse JSON easily.

I wont publish it as release  for now but you can use this

closed 2023-11-14 by webspiderteam