webspiderteam/Bluetooth-DMM-For-Windows #63: OBS Windows capture

sandro197802, 2025-09-01:
Hi all,
I'm trying to capture the Bluetooth DMM Windows in OBS studio software. It doesn’t capture the reading, only a black Windows that shows the mouse trace when it goes over it. Is there any solution?

webspiderteam, 2025-09-02:
Hi. dtry this solution if you didn't try yet. in OBS Studio, go to window capture that you creaated for dmm app. Right click and select properties, then go to capture method, then click on from automatic which is enabled by default then go to Windows Graphics capture.  i hope that should fix it.

webspiderteam, 2025-09-02:
Youre welcome :) İ hope we can see your streams on youtube :)  This makes me happy when i see someone using it :) BTW your dmm can have some issues while decoding data because i can't find anyone that tests with software compeletely. if you see some weird readings on app inform me pls.

webspiderteam, 2025-09-02:
Than you very much 👍

closed 2025-09-07 by webspiderteam