webspiderteam/Bluetooth-DMM-For-Windows #64: UNI-T UT60BT

taidjn, 2025-09-07:
Hi,

First of all, thanks for creating this tool. It is what I want to use with my UNI-T UT60BT; however, it doesn't seem to work properly for me. 

Crashlog:

> error catch Error: System.Windows.Threading.DispatcherUnhandledExceptionEventArgs
An application error occurred.
Please check whether your data is correct and repeat the action. If this error occurs again there seems to be a more serious malfunction in the application, and you better close it.

Error: Object reference not set to an instance of an object.

Do you want to continue?
(if you click Yes you will continue with your work, if you click No the application will close)

webspiderteam, 2025-09-07:
can you try with this one 

[Debug.zip](https://github.com/user-attachments/files/22194970/Debug.zip)

taidjn, 2025-09-07:
> do you have discord?

Sure. Where can I join? Thank you.

webspiderteam, 2025-09-07:
you can find me as webspiderteam_71418

webspiderteam, 2025-11-15:
İt's already compatible but it has some problem in startup sequence. I made some versions for solving this issue but not completely get feedback.

[debugrequestfirst.zip](https://github.com/user-attachments/files/23557485/debugrequestfirst.zip)
[degugforcedaftertype.zip](https://github.com/user-attachments/files/23557486/degugforcedaftertype.zip)
İf you can test and give me feedback i can make it as a stable version